sandialabs/WecOptTool #299: Errors running tutorials

jwills801, 2023-11-21:
**Describe the bug**
I am very new to WecOptTool, so I'm not sure if this is a bug or if I just need help. In the tutorial files (at least in the tutorials 1 and 2), there is an error when defining the wec object. The lines defining stiffness and inertia have been removed and the code cannot run. In previous versions and in the documentation, there are additional lines defining the wec stiffness and inertia.

**To Reproduce**
Steps to reproduce the behavior:
1. Run tutorial 1 or 2
3. Scroll down to where the wec object is defined
4. See error

**Expected behavior**
Maybe a warning about negative damping values

**Observed behavior**
[image]

**System:**
 - Ubuntu 20.04
 - Python version: 3.11.5
 - WecOptTool version: 2.6.0

cmichelenstrofer, 2023-11-21:
The issue here is that the GitHub repo is ahead of the release versions and we have introduced some breaking changes. We currently point users to the tutorials on GitHub which can lead to this problem... We are discussing possible solutions. 

@jwills801 for now get the tutorials from the latest release (tag `v2.6.0`) [here](https://github.com/sandialabs/WecOptTool/tree/v2.6.0). Hopefully those work for you!

cmichelenstrofer, 2023-11-22:
After releasing `v3.0` we will change our workflow to use a `dev` branch. All pull requests should be `dev`. The `dev` branch will only be merged into `main` when we are releasing a new version. This will guarantee `main` is at the same place as the latest release that can be obtained through `pip` or `conda`/`mamba`.

jtgrasb, 2024-04-01:
As indicated by the comment above, we switched to a workflow with a `dev` and a `main` branch. The `dev` branch will be used for all development efforts while the `main` branch will remain consistent except when new versions are released. This will guarantee `main` is at the same place as the latest release that can be obtained through `pip` or `conda`/`mamba`.

closed 2024-04-01 by jtgrasb

ryancoe, 2024-04-02:
@jtgrasb - Maybe we should 

1. add something to [`CONTRIBUTING.md`](https://github.com/sandialabs/WecOptTool/blob/main/.github/CONTRIBUTING.md) and/or 
2. make it so that new pull requests are automatically created against the `dev` branch (I've been searching on this and I don't see a way to do it -- lots of other people asking for this capability though...)?

https://github.com/sandialabs/WecOptTool/blob/bdf192c8810895e0cd711c306a9fedb65b75e670/.github/CONTRIBUTING.md?plain=1#L41-L45