sandialabs/WecOptTool #303: BUG: `from_impedance` index error for numpy array

rebeccamccabe, 2023-12-23:
**Describe the bug**
WecOptTool tries to index a string into the excitation coeff when running optimization with a wec created with the `from_impedance` method, which produces an index error when `exc_coeff` is a numpy array.

**To Reproduce**
Steps to reproduce the behavior:
Minimum working example of an optimization where impedance and excitation are defined with numpy arrays.
```
import autograd.numpy as np
import wecopttool as wot

m = np.array([1.0])
w = np.array([1.0])
F_h = np.array([1.0])
B_h = np.array([1.0])
K_h = np.array([1.0])
impedance = m * 1j * w + B_h + K_h/(1j * w)

ndof = 1
nfreq = 1

impedance = np.reshape(impedance,(ndof,ndof,nfreq))
F_h = np.reshape(F_h,(ndof,nfreq))
K_h = np.reshape(K_h,(ndof,ndof))

name = ["PTO_Heave",]
kinematics = np.eye(ndof)
controller = None
loss = None
pto_impedance = None
pto = wot.pto.PTO(ndof, kinematics, controller, pto_impedance, loss, name)

f_add = {'PTO': pto.force_on_wec}

f1 = w / (2 * np.pi) # Hz
wavefreq = w / (2 * np.pi)
amplitude = 1
phase = 0
wavedir = 0
waves = wot.waves.regular_wave(f1, nfreq, wavefreq, amplitude, phase, wavedir)

wec = wot.WEC.from_impedance(
    freqs=wavefreq,
    impedance=impedance,
    exc_coeff=F_h,
    hydrostatic_stiffness=K_h,
    constraints=None,
    f_add=f_add
)

obj_fun = pto.mechanical_average_power
nstate_opt = 2*nfreq

options = {'maxiter': 200}
scale_x_wec = 1e1
scale_x_opt = 1e-3
scale_obj = 1e-2

results = wec.solve(
    waves,
    obj_fun,
    nstate_opt,
    optim_options=options,
    scale_x_wec=scale_x_wec,
    scale_x_opt=scale_x_opt,
    scale_obj=scale_obj,
)
```

**Expected behavior**
I expected this to run without error because the [documentation](https://sandialabs.github.io/WecOptTool/api_docs/wecopttool.WEC.from_impedance.html#wecopttool.WEC.from_impedance) indicates that `exc_coeff` should be `ArrayLike`.

**Observed behavior**
The line `exc_coeff['omega'].values` seems to be assuming `exc_coeff` is a dict, not an array.

```---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Untitled-1.ipynb Cell 1 line 5
     [48](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=47) scale_x_opt = 1e-3
     [49](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=48) scale_obj = 1e-2
---> [51](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=50) results = wec.solve(
     [52](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=51)     waves,
     [53](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=52)     obj_fun,
     [54](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=53)     nstate_opt,
     [55](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=54)     optim_options=options,
     [56](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=55)     scale_x_wec=scale_x_wec,
     [57](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=56)     scale_x_opt=scale_x_opt,
     [58](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=57)     scale_obj=scale_obj,
     [59](vscode-notebook-cell:Untitled-1.ipynb?jupyter-notebook#W0sdW50aXRsZWQ%3D?line=58) )

File [c:\Users\rgm222\Anaconda3\envs\wecopttool\lib\site-packages\wecopttool\core.py:805](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:805), in WEC.solve(self, waves, obj_fun, nstate_opt, x_wec_0, x_opt_0, scale_x_wec, scale_x_opt, scale_obj, optim_options, use_grad, maximize, bounds_wec, bounds_opt, callback)
    [802](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:802)     problem['jac'] = grad(obj_fun_scaled)
    [804](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:804) # minimize
--> [805](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:805) optim_res = minimize(**problem)
    [807](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:807) msg = f'{optim_res.message}    (Exit mode {optim_res.status})'
    [808](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:808) if optim_res.status == 0:

File [c:\Users\rgm222\Anaconda3\envs\wecopttool\lib\site-packages\scipy\optimize\_minimize.py:705](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/scipy/optimize/_minimize.py:705), in minimize(fun, x0, args, method, jac, hess, hessp, bounds, constraints, tol, callback, options)
    [702](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/scipy/optimize/_minimize.py:702)     res = _minimize_cobyla(fun, x0, args, constraints, callback=callback,
    [703](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/scipy/optimize/_minimize.py:703)                             **options)
...
-> [2166](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:2166) omega_e = exc_coeff['omega'].values
   [2167](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:2167) dir_w = waves['wave_direction'].values
   [2168](file:///C:/Users/rgm222/Anaconda3/envs/wecopttool/lib/site-packages/wecopttool/core.py:2168) dir_e = exc_coeff['wave_direction'].values

IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices
```

**System:**
 - OS: windows 10
 - Python version: 3.9.18
 - WecOptTool version: 2.6.0

**Additional information**
From #300 it seems like instead of a np array, an xarray is used, so I will try that for now.

rebeccamccabe, 2023-12-23:
I looked at old issues and it seems like this is a duplicate of #183, and the docstrings have been updated with #248 on `main` but not on v2.6.0 docs which is what I was looking at on the website.

jtgrasb, 2024-01-09:
Commenting to confirm that this error is solved with PR #248. The docs will be automatically updated when a new version is released. I am leaving this open for now to facilitate discussion on making sure our docs are consistent with the available source code in the future.

closed 2024-01-11 by jtgrasb