sandialabs/WecOptTool #304: BUG: docstrings have wrong arguments for `wec.post_process`

rebeccamccabe, 2023-12-24:
**Describe the bug**
Docstrings in `wec.post_process`, `pto.post_process`, and `wec.solve` should be updated to use the proper arguments. WEC is used twice, creating an error.

**To Reproduce**
`res_wec_fd, res_wec_td = wec.post_process(wec,results)` (errors)
`res_wec_fd, res_wec_td = wec.post_process(results,waves)` (ok)

**Expected behavior**
The suggested usage in the docstrings should run without error.

**Observed behavior**
```
Exception has occurred: AttributeError
'WEC' object has no attribute 'x'
  File "C:\Users\rgm222\Documents\Github\SEA-Lab\WEC-DECIDER\force_saturation_sweep.py", line 148, in inner_function
    res_wec_fd, res_wec_td = wec.post_process(wec,results)
  File "C:\Users\rgm222\Documents\Github\SEA-Lab\WEC-DECIDER\force_saturation_sweep.py", line 171, in sweep_nondim_coeffs
    X[idx] = inner_function(zeta_u_mat.ravel()[i], w_u_star_mat.ravel()[i], f_max_Fp_mat.ravel()[i], m, w, F_h, wavefreq = w/(2*np.pi), amplitude = 1)
  File "C:\Users\rgm222\Documents\Github\SEA-Lab\WEC-DECIDER\force_saturation_sweep.py", line 188, in <module>
    sweep_nondim_coeffs()
AttributeError: 'WEC' object has no attribute 'x'
```

**System:**
 - OS: windows 10
 - Python version: 3.9.12
 - WecOptTool version: 2.6.0

cmichelenstrofer, 2024-01-08:
the examples (not just the list of parameters) are wrong.

jtgrasb, 2025-01-02:
Closing this issue as it was resolved by #337.

closed 2025-01-02 by jtgrasb