sandialabs/WecOptTool #305: Optimization takes much longer with autograd

rebeccamccabe, 2023-12-26:
Running with `use_grad=False` (disabling autograd) speeds up the solve by a factor of around four, which is unexpected because I thought the point of using autograd is it's supposed to be quicker than the repeated function evaluations with finite difference. Apparently, a single autograd evaluation takes ~4x longer than the 20 function evaluations required for a single finite difference evaluation.

I'm running with a wec defined from impedance, and no extra user-defined code that might influence the derivative (`f_add` is just `pto.force_on_wec`, and the only constraint is maximum `pto.force_on_wec`). Regular wave, `nfreq = 5` and `nsubsteps = 16`.

`use_grad=False` results:
```
Optimization terminated successfully    (Exit mode 0)
            Current function value: -0.08664520621620488
            Iterations: 10
            Function evaluations: 211
            Gradient evaluations: 10
Time elapsed: 0.42539677796540437
```
`use_grad=True` results:
```
Optimization terminated successfully    (Exit mode 0)
            Current function value: -0.08664520621384908
            Iterations: 10
            Function evaluations: 11
            Gradient evaluations: 10
Time elapsed: 1.8151584642904777
```
Any idea what is going on?

rebeccamccabe, 2024-01-02:
I may have figured out what is going on. The autograd jacobian probably uses reverse mode, which is efficient for functions with many inputs and a single / few outputs. Because I had `nsubsteps` set high (16), there were many outputs, and reverse mode was inefficient and forward mode would probably work better. I tried enabling autograd on another problem where `nsubsteps=1`, and it was faster by a factor of ~2 than with autograd disabled. 

Maybe, based on the number of constraints and timesteps (outputs), and number of frequencies and x_opt variables (inputs), WecOptTool could decide to use forward or reverse mode as most appropriate.

cmichelenstrofer, 2024-01-08:
This is because we are using automatic differentiation for all of these: 
- the objective function: many-to-one function; main optimization (gradient descent).
- the dynamic constraints: many-to-many function; both input and output sizes scale with `nfreq` and `ndof`
- any other constraint: case-specific output size. Always many-inputs

Suggested test:
- Run the tutorials using autograd only for the objective function. See if the results are the same and if performance improves. 

If this is successful, suggested solution:
- Add a separate flag for using autograd for the dynamic equality. 
- leave it up to the users to add autograd to their additional `constraints` dictionary.

cmichelenstrofer, 2024-04-02:
I think this was just slower because of using `nsubsteps` in the solution. `nsubsteps` should be used only in the post-processing. But if this comes up again we will consider adding the ability to turn autograd on/off separately for the objective function vs the constraints. Closing for now.