sandialabs/WecOptTool #309: BUG: Velocity and acceleration inconsistent, despite small residual

rebeccamccabe, 2024-01-01:
**Describe the bug**
There are some combinations of inputs (I haven't been fully systematic but it seems like it happens when there are no user defined constraints and I'm optimizing for mechanical not electrical power) that create strange high frequency oscillations in the force and acceleration signals, while the position and velocity signals do not have these oscillations. The existence of the spiky oscillations is a minor annoyance, but the bigger problem is that the velocity and acceleration are inconsistent (derivative condition not adequately enforced), even though the residual to the dynamics equality constraint is quite small.

**To Reproduce**
```python
nfreq = 11
w = np.array( [1.] )
ws = np.arange(w, w*(nfreq+1), w)
B_h = 1/3 * ws
K_h = np.array( [25/9 ])
name = ["PTO_Heave",]
kinematics = np.eye(ndof)    
controller = None
f_max = None
    
def const_f_pto(wec, x_wec, x_opt, waves):
    f = pto.force_on_wec(wec, x_wec, x_opt, waves, nsubsteps)
    return f_max - np.abs(f.flatten())
ineq_cons1 = {'type': 'ineq',
              'fun': const_f_pto,
              }
constraints = [ineq_cons1] if f_max is not None else None

nstate_opt = 2*nfreq

# analytical impedance match solution
K_p = m * w**2 - K_h
B_p = B_h
Fp_phase = np.arctan(B_p*ws / K_p)
real_part_Fp  = -Fp * np.cos(Fp_phase)
imag_part_Fp  = -Fp * np.sin(Fp_phase)
x_opt_0 = np.concatenate([real_part_Fp,imag_part_Fp])
    
loss = None
pto_imp =  np.array([[0,-1],[-1,0.01]])
pto_impedance = np.tile(np.expand_dims(pto_imp,2), [1,1,nfreq])
pto = wot.pto.PTO(ndof, kinematics, controller, pto_impedance, loss, name)

f_add = {'PTO': pto.force_on_wec}

# define impedance and reshape
impedance = m * 1j * ws + B_hs + K_h/(1j * ws)
impedance = np.reshape(impedance,(nfreq,ndof,ndof))
K_h = np.reshape(K_h,(ndof,ndof))

exc_coeff, impedance = make_xarrays(ws, F_h, impedance)

wec = wot.WEC.from_impedance(
    freqs=freqs,
    impedance=impedance,
    exc_coeff=exc_coeff,
    hydrostatic_stiffness=K_h,
    constraints=constraints,
    f_add=f_add
)

phase = 30
wavedir = 0
waves = wot.waves.regular_wave(f1, nfreq, wavefreq, amplitude, phase, wavedir)
wave = waves[:,:,0]

obj_fun = pto.mechanical_average_power

# use the analytical unsaturated solution as a guess
dc_pos = np.array([0])
zeta = B_h / (m * ws)
X_unsat = F_h[0] / (m * w**2) / (2*zeta)
real_part_pos = np.full_like(ws, X_unsat)
imag_part_pos = np.zeros_like(real_part_pos)
x_wec_0 = np.concatenate([dc_pos, real_part_pos, imag_part_pos[:-1]])

results = wec.solve(
    waves,
    obj_fun,
    nstate_opt,
    optim_options={'maxiter': 200},
    scale_x_wec=1,
    scale_x_opt=1,
    scale_obj=1,
    x_wec_0=x_wec_0,
    x_opt_0=x_opt_0,
    use_grad=False
)

x_wec, x_opt = wec.decompose_state(results[0].x)
r = wec.residual(x_wec, x_opt, wave)
print('Residual: ',r)

res_wec_fd, res_wec_td = wec.post_process(results[0],wave,nsubsteps=nsubsteps)
res_pto_fd, res_pto_td = pto.post_process(wec,results[0],wave,nsubsteps=nsubsteps)

plt.figure()
res_wec_td.pos.plot()
res_wec_td.vel.plot()
res_wec_td.acc.plot()
res_pto_td.force.plot()
res_pto_td.power.sel(type='elec').plot.line('--',x='time')
plt.legend([res_wec_td.pos.long_name, res_wec_td.vel.long_name, 
            res_wec_td.acc.long_name, res_pto_td.force.long_name,
            res_pto_td.power.long_name])
```

**Observed behavior**
Run code as-is: 
[image]
```
Optimization terminated successfully    (Exit mode 0)
            Current function value: -0.34148166179864475
            Iterations: 31
            Function evaluations: 1396
            Gradient evaluations: 31
Residual:  [-5.98854299e-13  1.44106949e-11  2.20581331e-11 -9.12470099e-12
 -1.89053218e-11  1.96127559e-11 -1.84963156e-12  1.39777079e-12
  3.53717056e-13 -9.44588852e-12  1.46865853e-11 -2.42028619e-12
 -2.64233080e-12  3.15547588e-12  1.50990331e-12 -5.15094634e-11
  1.15445431e-11 -1.85269577e-11 -3.07889270e-11 -2.50555132e-12
  4.68647343e-12  2.82340817e-12]
```
The optimization converges and the residual is small, but the discrepancy between velocity and acceleration is a problem.
I also plotted the three component forces, and the excitation force is smooth, while the PTO and intrinsic impedance forces have the spikes.

Run code with `obj_fun = pto.average_power`:
[image]
So optimizing for electrical power apparently solves the problem.

Run code with `f_max = np.array( [0.94] )`
[image]
So adding a force limit also apparently solves the problem.

**Expected behavior**
When optimizing only for mechanical power with no force limit, I think that it is actually perhaps expected to have high frequency force harmonics in the PTO force, because there is nothing in the objective preventing the controller from applying them (they do not affect power). They would only go away when optimizing for electrical power, since the extra unnecessary motion would add dissipation and reduce power, or when there is a force constraint since the extra amplitude would violate the constraints. So the problem is not the presence of the spikes, but the fact that the spikes are not present in the velocity and position waveforms, indicating that the `dv/dt = a` portion of the dynamics is not being properly enforced, while `F=ma` seems to be fine. 

Separately, if we wanted to get rid of the spikes, perhaps it makes sense to add a tiny norm of PTO force harmonics term to the `pto.mechanical_average_power` objective function, scaled such that it will have negligible effect on results and just will ensure that the additional high frequency harmonics don't occur when optimizing mechanical power. I did observe that the frequency of the spikes increases when I increase `nfreq`, so maybe it is only the highest harmonic where these oscillations occur.

**System:**
 - OS: Windows 10
 - Python version: 3.11.7
 - WecOptTool version: `main` branch at 598e875cd48751ddb652657391205f6e208714f0

cmichelenstrofer, 2024-01-08:
The residual (dynamic equations) is small, but the dynamics are wrong. I am thinking is this an issue with scaling. 

Possible solution: 
- **change the residual output to be unscaled**.

dtgaebe, 2024-01-08:
The spikiness originates from the imaginary part of the last frequency component...
I don't know why the solver converges to have content there in x_opt.
[image]
When artificially removing the very last component we get what we'd expect:
```python
x_opt[-1] = 0
x_wec[-1] = 0

res_artificial  = results[0]
res_artificial.x = np.concatenate([x_wec, x_opt])

res_wec_fd, res_wec_td = wec.post_process(res_artificial,wave,nsubsteps=nsubsteps)
res_pto_fd, res_pto_td = pto.post_process(wec,res_artificial,wave,nsubsteps=nsubsteps)

plt.figure()
res_wec_td.pos.plot()
res_wec_td.vel.plot()
res_wec_td.acc.plot()
res_pto_td.force.plot()
res_pto_td.power.sel(type='elec').plot.line('--',x='time')
plt.legend([res_wec_td.pos.long_name, res_wec_td.vel.long_name, 
            res_wec_td.acc.long_name, res_pto_td.force.long_name,
            res_pto_td.power.long_name])
```
[image]

cmichelenstrofer, 2024-01-08:
might be related to #225?

cmichelenstrofer, 2024-04-02:
This is not a bug. If you use average mechanical power as the objective function, an unstructured controller, and no constraints, then the Nyquist component of the PTO force and the WEC position is arbitrary. This is the high frequency component you are observing. If you re-run this case you should see a different magnitude for it each time. The Nyquist component of the velocity is zero which is why you don't see the high frequency oscillations. 

@dtgaebe and I wrote this up showing that these components are arbitrary in this case: [Power_last_frequency.pdf](https://github.com/sandialabs/WecOptTool/files/14841972/Power_last_frequency.pdf). We are also documenting this in #225 and closing it.

rebeccamccabe, 2024-04-03:
> The Nyquist component of the velocity is zero which is why you don't see the high frequency oscillations.
Does this mean that acceleration is not expected to match the derivative of velocity? If I want them to match, would I need to redefine the derivative matrix so that the second derivative is phase shifted 90°?

cmichelenstrofer, 2024-04-03:
Correct, the acceleration does not match the derivative of the velocity at the Nyquist frequency. There is no way of making position-velocity-acceleration consistent at the Nyquist frequency. However if you truncate your Fourier expansion at a high enough frequency, the Nyquist component should not be significant. The problem here is that for this specific problem the Nyquist component is arbitrary and can be quiet large. The best option if you want to use average mechanical power as the objective and an unstructured controller might be to constraint the Nyquist components to be equal to zero. Here is more info on [FFT-based differentiation](https://math.mit.edu/~stevenj/fft-deriv.pdf).