sandialabs/WecOptTool #311: Feature request: Additional error checking

rebeccamccabe, 2024-01-02:
**Feature description.**
Add error checking for the size of the x0 initial guess and the matching of frequency data.

**Issue addressed.**
In my use of WOT the past few weeks, I was initially doing some things wrong (user error, nothing wrong with WOT), and my debugging process was more time consuming due to lack of some specific error messages in WOT. So it would be great to help future users who may make the same mistakes as I did.

**Describe the solution you'd like**
For x0, insert the following lines in `wec.solve`, before the line `x0 = np.concatenate([x_wec_0, x_opt_0])*scale`:
```python
assert np.shape(x_wec_0) == (self.nstate_wec,)
assert np.shape(x_opt_0) == (nstate_opt,)
```

For frequency data, right now in `wave_excitation`, it enforces `np.allclose(omega_w, omega_e)` (equal frequency values for excitation and waves xarrays). However, nowhere does it enforce whether the `freqs` passed to the wec object itself matches either of these. I was initially confused by which units were rad/s vs Hz (more documentation on this might be useful). By my current understanding, the excitation and impedance xarrays must be constructed using rad/s, the waves xarray must be constructed using Hz, and the wec object must be constructed using Hz. I initially (incorrectly) constructed the wec object with rad/s instead of Hz (which did not match my correctly created xarrays), and there was no error checking preventing this. I am not certain on the best place for this error check to go, ie if doing it in `wave_excitation` is sufficient.

ryancoe, 2024-01-02:
Good suggestions @rebeccamccabe. We'll discuss this as a team later this week. I think we can consider two distinct questions here:

1. **Should we be validating arguments more than we are?** ...specifically on the initial guesses for the decision variable, which I agree can be a bit tricky to remember what the sizes should be. We've generally taken what I believe could be called a ["duck typing" approach](https://ioflood.com/blog/duck-typing/), where we don't do much argument validation, instead assuming that errors will be thrown when appropriate _and the user will be able to understand them._
2. **rad/s vs. Hz** - sounds like we're requiring a mixture of these units for different inputs, that's definitely not great...

cmichelenstrofer, 2024-01-08:
The difference in units comes from an early decision to "use degrees and Hz for user facing input/output and radians and rad/s for internal calculations". This was OK when we use Capytaine but when we added `from_impedance` this brakes... I see two options:

1. change `from_impedance` to accept `Hz` instead.
2. just choose one (dictated by capytaine) and use it consistently *everywhere*

ryancoe, 2024-01-11:
Let's check whether `wavespectra` uses `rad/s` or `Hz`

ryancoe, 2024-04-03:
@rebeccamccabe - Upon further review...

- additional error checking: We're not going to take any action on this at the moment for the reasons stated above. We are considering some substantial restructuring that might help with this in the future.
- frequency units: see #331

closed 2024-04-03 by ryancoe