sandialabs/WecOptTool #314: DOCUMENTATION: Improve explanation of `nsubsteps` usage

michaelcdevin, 2024-01-09:
Derived from #307.

To my understanding, WecOptTool currently uses substeps for two purposes:

1. To enforce constraints at more frequencies than needed for the dynamics equation without excessive computational expense (see https://github.com/sandialabs/WecOptTool/issues/307#issuecomment-1881785357)
2. To discretize results for smoother plotting for post-processing

The documentation is rather indirect about these two uses: Tutorial 1 [broadly points to](https://github.com/sandialabs/WecOptTool/blob/598e875cd48751ddb652657391205f6e208714f0/examples/tutorial_1_WaveBot.ipynb#L200) the Theory documentation for an explanation of using substeps in constraints without specifying where, and then [is a bit handwavy](https://github.com/sandialabs/WecOptTool/blob/598e875cd48751ddb652657391205f6e208714f0/examples/tutorial_1_WaveBot.ipynb#L345C1-L346C188) for its use for post-processing. The Theory documentation does include some explanation [in the Constraints section](https://github.com/sandialabs/WecOptTool/blob/598e875cd48751ddb652657391205f6e208714f0/docs/source/theory.rst?plain=1#L108-L110)):

> An important practical factor when using this functionality is to make sure that the constraint is evaluated at a sufficient number of collocation points. It may be required to enforce constraints at more points than the dynamics (as defined by the frequency array).
In WecOptTool's example PTO module, this is controlled by the `nsubsteps` argument (see, e.g. `wecopttool.pto.PTO.force_on_wec`).

This could probably be improved, as "it may be required" could be changed to explain why you would want to use substeps specifically for the constraints instead of changing the frequency vector.

Overall, I think the use of `nsubsteps` could be clarified in a number of ways:

* The reference in Tutorial 1 to the Theory documentation should be made into a hyperlink going directly to the Constraints section (I believe this used to be the case but was changed due to CI issues with Sphinx `linkcheck`, but this is not a concern since this was removed in #xxx)
* The explanation in the Theory documentation could be made more specific, i.e. explain _why_ "it may be required" to use substeps for the constraints
* The difference in using `nsubsteps` for the constraints and for post-processing could be clarified, either by providing language for this in Tutorial 1, or perhaps by changing the variable name to something else for one of these, since it is two rather distinct uses for the same variable name

ryancoe, 2024-01-10:
Per discussion yesterday with @michaelcdevin @cmichelenstrofer @dtgaebe @gbacelli, will also be good to make clear the distinction between `nsubsteps` and `nfreq` (inequality constraints vs. equality constraints) -- related to #307.

barrosyan, 2026-10-09:
I would like to take this documentation issue. Plan: (1) expand the Constraints section of the Theory page to explain why inequality constraints may need more evaluation points than the dynamics, what `nsubsteps` does for them (finer evaluation of the same Fourier series, no new decision variables) and how that differs from increasing `nfreq`; (2) link Tutorial 1 directly to that section (the current link has doubled parentheses); (3) state in Tutorial 1 that `nsubsteps` in `post_process` only refines the output time series. I will not rename any argument.

dtgaebe, 2026-10-09:
@barrosyan thank you for bringing this up. We have improved the theory on `nfreq` slightly [here](https://sandialabs.github.io/WecOptTool/theory.html#time-frequency-equivalence-in-the-pseudo-spectral-method ), but never worked on explaining `nsubsteps` further.