sandialabs/WecOptTool #317: Real component of PTO impedance must be constant for all frequencies Error

degoeden, 2024-02-07:
Hi,

I was trying to create a PTO with an impedance that has a real component that is dependent on frequency when I ran into this error. Upon reading the error message and looking in the code, it makes a lot of sense that I am receiving the error, but I'm confused as to why this check is present. It seems to limit my PTO circuit design to only use components in series so that I can use KVL (I was trying to model a circuit that includes elements in parallel which forces me to use KCL, which forces me to have real components that depend on frequency). 

I disabled the check for now (unfortunately I can't say if it works or not yet because I have been running into another error) but was wondering if you could tell me why this check is there, and if it will be problematic that I disabled it. Thank you!

dtgaebe, 2024-02-07:
Hi Nate,

I think for now it is good that you disabled the check. If I remember correctly we implemented the check because we anticipated only PTO impedance with constant loss (real part) due the linear nature. 
However, we very well might be wrong.

Do you want to share more about the PTO you're trying to model? Feel free to do this offline.

ryancoe, 2024-02-08:
For reference, here's the check that was giving @degoeden trouble. The decision to include this related to how we assign the `f=0` elements of the PTO impedance... current thinking is that this check should maybe be removed.

https://github.com/sandialabs/WecOptTool/blob/528c679b0e6f393da30c7e0523356a99e6b34ed1/wecopttool/pto.py#L137-L144

dtgaebe, 2024-04-03:
To cross-reference a related issues:
https://github.com/sandialabs/WecOptTool/issues/309#issuecomment-2032712699
and #225 

Long story short: The PTO impedance at `f=0` makes the optimization problem well defined for the DC component of unstructured controller when optimizing for electrical power