sandialabs/WecOptTool #318: Installation errors

Khalid-1994-06-17, 2024-02-19:
Hello my name is Khalid Al Nahyan,
I am a student studying his PhD in MBZUAI, I came with is software that I would like to use as a part of my PhD research topic but when I tried to install it on my MacBook with the M2Pro chip and my iMac Pro with Intel chip using pip, I got the following error:

Collecting wecopttool
  Downloading wecopttool-2.6.0-py3-none-any.whl.metadata (4.1 kB)
Requirement already satisfied: numpy>=1.20 in /Users/khalidnahyan/anaconda3/lib/python3.10/site-packages (from wecopttool) (1.26.2)
Requirement already satisfied: scipy in /Users/khalidnahyan/anaconda3/lib/python3.10/site-packages (from wecopttool) (1.11.4)
Requirement already satisfied: xarray in /Users/khalidnahyan/anaconda3/lib/python3.10/site-packages (from wecopttool) (2022.11.0)
Collecting autograd (from wecopttool)
  Using cached autograd-1.6.2-py3-none-any.whl.metadata (706 bytes)
Collecting capytaine (from wecopttool)
  Using cached capytaine-2.0.tar.gz (352 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [27 lines of output]
      + meson setup /private/var/folders/55/wmhf0mt132d97tjq7jyn861h0000gn/T/pip-install-s79foa5j/capytaine_4d9fefa8a583418a92a85c6702778402 /private/var/folders/55/wmhf0mt132d97tjq7jyn861h0000gn/T/pip-install-s79foa5j/capytaine_4d9fefa8a583418a92a85c6702778402/.mesonpy-0t9bil4c -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/private/var/folders/55/wmhf0mt132d97tjq7jyn861h0000gn/T/pip-install-s79foa5j/capytaine_4d9fefa8a583418a92a85c6702778402/.mesonpy-0t9bil4c/meson-python-native-file.ini
      The Meson build system
      Version: 1.3.2
      Source dir: /private/var/folders/55/wmhf0mt132d97tjq7jyn861h0000gn/T/pip-install-s79foa5j/capytaine_4d9fefa8a583418a92a85c6702778402
      Build dir: /private/var/folders/55/wmhf0mt132d97tjq7jyn861h0000gn/T/pip-install-s79foa5j/capytaine_4d9fefa8a583418a92a85c6702778402/.mesonpy-0t9bil4c
      Build type: native build
      Project name: capytaine
      Project version: 2.0
      
      ../meson.build:1:0: ERROR: Unknown compiler(s): [['gfortran'], ['flang'], ['nvfortran'], ['pgfortran'], ['ifort'], ['ifx'], ['g95']]
      The following exception(s) were encountered:
      Running `gfortran --version` gave "[Errno 2] No such file or directory: 'gfortran'"
      Running `gfortran -V` gave "[Errno 2] No such file or directory: 'gfortran'"
      Running `flang --version` gave "[Errno 2] No such file or directory: 'flang'"
      Running `flang -V` gave "[Errno 2] No such file or directory: 'flang'"
      Running `nvfortran --version` gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running `nvfortran -V` gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running `pgfortran --version` gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running `pgfortran -V` gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running `ifort --version` gave "[Errno 2] No such file or directory: 'ifort'"
      Running `ifort -V` gave "[Errno 2] No such file or directory: 'ifort'"
      Running `ifx --version` gave "[Errno 2] No such file or directory: 'ifx'"
      Running `ifx -V` gave "[Errno 2] No such file or directory: 'ifx'"
      Running `g95 --version` gave "[Errno 2] No such file or directory: 'g95'"
      Running `g95 -V` gave "[Errno 2] No such file or directory: 'g95'"
      
      A full log can be found at /private/var/folders/55/wmhf0mt132d97tjq7jyn861h0000gn/T/pip-install-s79foa5j/capytaine_4d9fefa8a583418a92a85c6702778402/.mesonpy-0t9bil4c/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Note: you may need to restart the kernel to use updated packages.

How could I resolve this issue and when I tried conda it will take a long time to install

Please give me some trouble shooting step by step

ryancoe, 2024-02-26:
@Khalid-1994-06-17 - Did this resolve your issue?

Khalid-1994-06-17, 2024-02-27:
Hi sorry for the late reply wecopttool is now working very well on both computers thanks. If there is anything else I will let you guys know