sandialabs/WecOptTool #320: Minor rendering problems in documentation

ryancoe, 2024-03-21:
There's a couple minor rendering bugs in the documentation, e.g., 

[image]

michaelcdevin, 2024-03-21:
Some other rendering problems I've found:

**Tutorial 1**:
* Lists not rendering in Section 1 
[image]
[image]

**Tutorial 4**:
* Lists not rendering in Sections 1.4 and 1.5
[image]
* Math not rendering right in Section 1.6 
[image]

michaelcdevin, 2024-10-18:
Resolved in #339

closed 2024-10-18 by michaelcdevin