sandialabs/WecOptTool #324: Install on an ARM-based mac

ryancoe, 2024-03-28:
_I have been through this a number of times, and always forget how I solved the problem, so maybe this is useful to someone else._

If you try the following:

```zsh
mamba create -n wot_tmp1
mamba activate wot_tmp1
mamba install -c conda-forge wecopttool
```

you will get

```zsh
Looking for: ['wecopttool']

conda-forge/osx-arm64                                       Using cache
conda-forge/noarch                                          Using cache
pkgs/r/noarch                                                 No change
pkgs/main/osx-arm64                                           No change
pkgs/main/noarch                                              No change
pkgs/r/osx-arm64                                              No change
Could not solve for environment specs
The following packages are incompatible
└─ wecopttool is not installable because there are no viable options
   ├─ wecopttool [2.0.1a1|2.1.0|2.2.0|2.2.2] would require
   │  └─ wavespectra, which does not exist (perhaps a missing channel);
   └─ wecopttool [2.2.2|2.2.3|...|2.6.0] would require
      └─ capytaine, which does not exist (perhaps a missing channel).
```

I think this is because `wavespectra` and `capytaine` both include compiled binaries, and `conda-forge` is not generating those for ARM processors (yet).

ryancoe, 2024-03-28:
Instead you should do the following:

1. Make sure you have compilers installed on your system (see discussion, e.g., here: https://github.com/capytaine/capytaine/issues/190)
2. Call the following commands (**note this version number for python is based on current version number that `wecopttool` needs per [`pyproject.toml`](https://github.com/sandialabs/WecOptTool/blob/main/pyproject.toml)**)

```zsh
mamba install "python<3.12" pip
pip install wecopttool\[geometry] jupyter
```

closed 2024-03-28 by ryancoe

ryancoe, 2024-03-28:
Note that if you want to develop, replace the last two commands with:

```zsh
mamba install "python<3.12" pip pytest ipykernel
pip install -e .\[dev,geometry]
```

cmichelenstrofer, 2024-04-17:
@ryancoe you cannot install `wavespectra` and `capytaine` from `conda`?

ryancoe, 2024-04-17:
[image]