sandialabs/WecOptTool #325: Simulate WEC without optimization or controller

brigittedg, 2024-03-29:
Hi, 
Before optimizing I want to simulate the dynamics of the WEC with the sea acting on it, without adding PTO force. This, to compare with results I optained using previous software, so no optimization or controller. Is there an easy way to do this?

So far using WecOptTool we have succesfully obtained the mesh, the various types of coefficients and the wave elevation spectrum.

I have tried various things so to simulate without PTO but they did not work, such as:
- Setting f_add to None or {}, however this resulted in a rank-deficiency error exit mode 7
- Making the objective function independent of the movement of the wec
- Joining individual forces from wot.standard_forces together to eventually divide by mass and added mass to obtain acceleration, velocity etc., but this seems like something I probably should be an easier way but I have not been able to find it.

michaelcdevin, 2024-03-29:
Hi @brigittedg,

It sounds like you're already on the right track with your second bullet point. The way we have done this in the past is to provide a dummy `x_opt` and objective function that solves immediately. `x_wec` is still forced to satisfy the residual constraint, so `WEC.solve()` still returns the proper dynamics of the WEC even though nothing is being optimized. Try, for example:
```
def obj_fun(wec, x_wec, x_opt, waves):
    return np.abs(x_opt)l
nstate_opt = 1
x_opt_0 = np.array([0])
results = wec.solve(
            waves,
            obj_fun,
            nstate_opt,
            x_opt_0=x_opt_0,
```

Setting `f_add=None` is valid, so it returning an error may indicate another problem with the model.

cmichelenstrofer, 2024-04-03:
Action: 

- [ ] Write a short documentation section explaining how to do this with a minimal working example

brigittedg, 2024-04-04:
@michaelcdevin Thank you very much! f_add now also did not have a problem anymore.

rebeccamccabe, 2024-06-18:
I read [this paper](https://ieeexplore.ieee.org/document/7950997) from professor Ringwood about a method called "nonlinear frequency domain method" instead of pseudospectral method, but I think it's the same thing just without optimizing. It seems to imply that when doing simulation without optimization, it is more efficient to solve with a nonlinear equation solver like Newton's method than it is to solve the optimization with a dummy objective. So perhaps there could be a separate simulation mode in WecOptTool that when x_opt is zero length then it uses Newton's method / another solver instead of SQP?