sandialabs/WecOptTool #338: Mesh is not inside the domain

riasatmorshed, 2024-04-18:
I am trying to run my lab-scale OSWEC which is of dimension width = 1.2m height = 0.7m and thickness = 0.2m. I have used the same geometry in NEMOH. However, while running the WecOptTool, despite using "keep_immersed_part()" I am getting the error: "WARNING:capytaine.bem.problems_and_results:The mesh of the body C:\Users\mdriasatmorshed\WEC-Sim\examples\BEMIO\NEMOH\rhino_parametric_const_area\a_1_longRange\Mesh\a_1.dat_immersed is not inside the domain.
Check the position of the free_surface and the water_depth
or use body.keep_immersed_part() to clip the mesh"

I am attaching two screenshots from my jupyter notebook for reference. 
[image]
[image]

I am kind of in a fix what is causing this problem even though I set the water depth to be 0.7m. 

Thanks!

ryancoe, 2024-04-18:
@riasatmorshed - Thanks for your interest! It looks to me like the issue is caused by the fact that your mesh is just below the fee surface (i.e., the upper surface of the flap is very close to `z=0`). There may be some issues in the [capytaine repo](https://github.com/capytaine/capytaine) about this (not finding anything immediately... maybe this one? https://github.com/capytaine/capytaine/issues/320). Anyhow, I think you can do one of the following things to resolve this problem:

1. Alter your mesh so that it extends above the free surface (i.e., set the flap `height` to be greater than the `depth`), so that it can be cleanly "clipped" when you call `keep_immersed_part`
2. If you really do want the flap to not pierce the free surface, shorten it sufficiently so that the problem goes away (for me, I was able to set the flap `height` to `0.9*depth` and that resolved the problem

<details>
  <summary>Example Python code</summary>

```python
import wecopttool as wot
import capytaine as cpy
import pygmsh
import gmsh

depth = 0.7
freq = wot.frequency(0.05, 20, False) # False -> no zero frequency

with pygmsh.occ.Geometry() as geom:
       gmsh.option.setNumber('Mesh.MeshSizeFactor', 1)
       box = geom.add_box([0, 0, -1*depth], 
                     [0.2, 0.8,  depth*0.9])
       mesh = geom.generate_mesh()
       
fb = cpy.FloatingBody.from_meshio(mesh, name="flap").keep_immersed_part()
fb.add_rotation_dof(cpy.Axis((0,1,0), (0,0,-1*depth)))

fb.show_matplotlib()

freq = wot.frequency(0.05, 20, False) # False -> no zero frequency
bem_data = wot.run_bem(fb, freq, 
                       depth=depth,
                       )

bem_data['added_mass'].squeeze().plot()
```

[image]
[image]

</details>

Please let us know if this resolves your problem.

riasatmorshed, 2024-04-21:
Hi @ryancoe! 

Thanks a lot for your prompt response! I am sorry for replying late. Your comment and code-snipped proved to be super useful and handy. Most importantly it solved the issue! Thanks again!

ryancoe, 2024-04-22:
Sure thing @riasatmorshed!

closed 2024-04-22 by ryancoe