sandialabs/WecOptTool #340: Scaling factor for the Optimization

riasatmorshed, 2024-04-23:
While I am trying to run my lab-scale OSWEC, which is of 1.2m x 0.2m x 0.7m. And the maximum diffraction force is around 70, Froude-Krylove force is around at the O(-5), maximum radiation damping is around 142- these are the data extracted from bem_data. 

Now, when I try to calculate the optimization loop with the following code: 
`%%time
options = {'maxiter': 200}
scale_x_wec = 1e1
scale_x_opt = 1e-1
scale_obj = 1e-2

results = wec.solve(
    waves, 
    obj_fun, 
    nstate_opt,
    optim_options=options, 
    scale_x_wec=scale_x_wec,
    scale_x_opt=scale_x_opt,
    scale_obj=scale_obj,
    )

opt_mechanical_average_power = results[0].fun
print(f'Optimal average mechanical power: {opt_mechanical_average_power} W')`

And my wave parameters are as follows: 

`amplitude = 0.15
wavefreq = 0.3 #in Hz
phase = 0
wavedir = 0
waves = wot.waves.regular_wave(f1, nfreq, wavefreq, amplitude, phase, wavedir)`

The program tends to take a long time, I am guessing which translates into that the solution is not converging. And I am suspecting it has something to do with scaling. Besides, I also read something about the scaling in a conversation among the developers. But I am not sure if I got the concept that they are discussing right. It would be great to have a second opinion on this code as to what might be going wrong. Thank you!

ryancoe, 2024-04-23:
@riasatmorshed - Would you be willing to post the complete code for your analysis? This would make it easier for us to provide assistance. Without running anything, I would guess that the issue has to do with one of the following areas.

1. Scaling: see, e.g., https://sandialabs.github.io/WecOptTool/theory.html#scaling
2. Number of frequencies used: see, e.g., https://sandialabs.github.io/WecOptTool/theory.html#frequencies

jtgrasb, 2024-04-24:
Hi @riasatmorshed, thank you for your inquiries. I have taken a look at your setup and have a few comments that I hope help clear up some of your questions:
1. Your original setup has a frequency array of 500 frequencies. This is much larger than we generally recommend. The optimization function (scipy.optimize.minimize), which is called when you run `wec.solve()`, uses 2 times the number of frequencies  as the number of optimization states. Thus, in this case, you would have 1000 optimization states that Scipy is trying to evaluate and optimize all at once. The required complexity of computations increase exponentially with the number of optimization states. Furthermore, many of your states (such as frequencies far from the wave excitation) likely do not contain any significant power and the optimizer is spending its resources on optimizing many equivalently arbitrary states. Both of these reasons lead to the 50 frequency simulation converging much quicker. If you look at our [tutorials](https://sandialabs.github.io/WecOptTool/tutorials.html), we usually use between 50 and 150 frequencies.
2. You need to be aware of the relationship between the frequency array and the wave conditions. Since you are using a wave with  a frequency of 0,3 Hz, that frequency needs to be included in your array. Thus, I would recommend starting with f1 = 0.3 Hz (no excitation or response will occur at any lower frequencies for regular waves). Then, for regular waves, the frequency array can be relatively small. When I run your example with f1 = 0.3 Hz and nfreq = 10, I can see the optimization runs smoothly and examine the results (using same plotting from tutorial 1): 
[image]
[image]
[image]
You can see that the PTO force never reaches its constraint (and you didn't actually apply the constraint when defining the WEC). I also recommend looking at the results in the frequency domain to get a better perspective of if all the response is being captured. Here's some code to get started:

```python
wec_fdom, wec_tdom = wec.post_process(results[0], waves.sel(realization=0), nsubsteps=nsubsteps)
pto_fdom, pto_tdom = pto.post_process(wec, results[0], waves.sel(realization=0), nsubsteps=nsubsteps)
excitation_force = wec_fdom['force'].sel(type='Froude_Krylov') + wec_fdom['force'].sel(type='diffraction')
position = wec_fdom['pos']
velocity = wec_fdom['vel']
pto_force =  pto_fdom['force']
power = pto_fdom['power'].sel(type='mech')

fig, ax = plt.subplots(5,1,figsize=(8, 12))

ax[0].stem(wec_fdom['freq'],abs(excitation_force))
ax[0].set_ylabel('Excitation')

ax[1].stem(wec_fdom['freq'],abs(position))
ax[1].set_ylabel('Position')

ax[2].stem(wec_fdom['freq'],abs(velocity))
ax[2].set_ylabel('Velocity')

# Plot PTO force
ax[3].stem(wec_fdom['freq'],abs(pto_force))
ax[3].set_ylabel('PTO Force')

# Plot power
ax[4].stem(wec_fdom['freq'],abs(power))
ax[4].set_ylabel('Power')
ax[4].set_xlabel('frequency [Hz]')
```
Produces the following plot. You can see the wave excitation, WEC response, PTO force, and power all only have one main component at 0.3 Hz. This is because without the PTO constraint impacting the response, the result is 100% linear. In the next plot, I used a much sctricter PTO constraint (100 N). You can see we get components of the response (especially PTO force and power) at odd harmonics of the exciting frequency. As you develop your model, I would recommend looking at the frequency domain plots continuously to make sure you are capturing the dynamics effectively and efficiently (especially if you use irregular waves, which you need to make sure are well captured by the frequency array as well as the WEC response). 
[image] [image]

riasatmorshed, 2024-04-25:
Thanks for the prompt response! I really appreciate you guys bearing with all the basic questions!

jtgrasb, 2024-04-25:
You're welcome! Closing this issue for now