sandialabs/WecOptTool #371: Gear ratio not taken into account for PTO kinematics

jtgrasb, 2024-10-03:
**Describe the bug**
The gear ratio not taken into account when calculating the PTO kinematics such as the position and velocity. For tutorial 1, I plotted the WEC position and PTO position side-by-side and they are the same. If we are saying a gear ratio of 12 is equivalent to a gearbox where 1 m = 12 rad of rotation, the PTO position should be 12 times greater than the WEC position.
[image]. 

I'm not sure the best way to resolve this yet. I think either the gear ratio should be included as part of the kinematics matrix or the impedance matrix should somehow factor into the kinematics calculations.

**To Reproduce**
Steps to reproduce the behavior:
1. Plot WEC position vs. PTO position in tutorials.

**Expected behavior**
PTO position should be equal to WEC position * gear ratio.

**Observed behavior**
N/A

**System:**
 - OS: Windows 11
 - Python version: 3.10.9
 - WecOptTool version: 2.6.0

**Additional information**
N/A

ryancoe, 2024-10-24:
@jtgrasb will think through and look for place(s) to add clarifying documentation

dtgaebe, 2025-03-27:
I recommend closing this issue. Looks like no issue to me.

The heave kinematics in heave are supposed to result in the linear PTO stroke (in meter), i.e. the PTO position. N*PTO stroke would be the _rotary PTO position_.

For the LUPA the constraining part is also the relative travel between spar and float, which we call max stroke in meters
https://github.com/sandialabs/WecOptTool/blob/cde6b775f3ae5248b0b71e13c0e3797cf6a0cf3b/examples/tutorial_3_LUPA.ipynb#L739

@jtgrasb, you're right, if you'd really want to constraint the shaft angle that this would require a formulation of the constraints that take the linear to rotary gear ratio into account, but for the LUPA this is not the case.
I could see example of rotary spring that do have end-stops as an application, but our current tutorials are fine.

closed 2025-03-27 by dtgaebe