sandialabs/WecOptTool #373: conf.py overwitten when you run build_docs.py

ryancoe, 2024-10-09:
It seems that the updates made in #347 to [build_docs.py](https://github.com/sandialabs/WecOptTool/blob/main/docs/build_docs.py) make it so that the latest version of [conf.py](https://github.com/sandialabs/WecOptTool/blob/main/docs/source/conf.py) is always used when building the docs (i.e., any changes you've made to [conf.py](https://github.com/sandialabs/WecOptTool/blob/main/docs/source/conf.py) will be overwritten). I noticed this when trying to use my workaround to avoid running the tutorials (see #335) while reviewing #372.

Any thoughts on this @michaelcdevin?

https://github.com/sandialabs/WecOptTool/blob/b9e1e0dd058723ba9d76f030c9c8264c1a2d8ca0/docs/build_docs.py#L56-L62

michaelcdevin, 2024-10-09:
I discovered this too while testing the exact same thing. I was able to work around the issue by:
1. Changing [versions.yaml](https://github.com/sandialabs/WecOptTool/blob/main/docs/versions.yaml) so `'latest'` is the branch I'm interested in testing (i.e. `new_bib_refs` for #372) and commenting out all other branches
2. Doing a temporary `git commit` with those changes along with the #335 change to conf.py
3. Running build_docs.py locally to test the changes
4. Reverting the changes when I'm done testing

This is obviously very clunky and not very practical. A better solution I was thinking of was adding input arguments to build_docs.py so e.g. it only performs the `git checkout` commands you highlighted if you specify `python build_docs.py --option production` or something. Should be a pretty straightforward change

michaelcdevin, 2024-10-17:
Resolved in #374/#375

PR #374 (2024-10-10): Add debug option for building documentation locally
## Description
Resolves #373. The default build option when calling `docs/build_docs.py` is `debug`, which only builds the documentation for the git branch you are currently on. Calling `python docs/build_docs.py --build production` will create multiple versions of the documentation as per #347, and is called in the GitHub CI workflows

## Type of PR
- [x] Bug fix
- [ ] New feature
- [x] Documentation
- [ ] Other: (specify)

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://github.com/sandialabs/WecOptTool/blob/main [... cut]
closed 2024-10-17 by michaelcdevin