sandialabs/WecOptTool #376: Update .gitignore for new Sphinx output location

ryancoe, 2024-10-11:
With #356 (and likely some underlying PR?), we started moving the HTML results from [build_docs.py](https://github.com/sandialabs/WecOptTool/blob/main/docs/build_docs.py) to `docs/pages/` in addition to `docs/_build`

https://github.com/sandialabs/WecOptTool/blob/b9e1e0dd058723ba9d76f030c9c8264c1a2d8ca0/docs/build_docs.py#L78

We should therefore update the following section of [.gitignore](https://github.com/sandialabs/WecOptTool/blob/b9e1e0dd058723ba9d76f030c9c8264c1a2d8ca0/.gitignore) to ignore the [build_docs.py](https://github.com/sandialabs/WecOptTool/blob/main/docs/build_docs.py) output.

https://github.com/sandialabs/WecOptTool/blob/b9e1e0dd058723ba9d76f030c9c8264c1a2d8ca0/.gitignore#L8-L11

**Action: update [.gitignore](https://github.com/sandialabs/WecOptTool/blob/b9e1e0dd058723ba9d76f030c9c8264c1a2d8ca0/.gitignore) so that we don't see all of the docs build artifacts showing up as git changes.**

michaelcdevin, 2024-10-17:
> With https://github.com/sandialabs/WecOptTool/pull/356 (and likely some underlying PR?)

FYI the "underlying PR" is #347, the move to `docs/pages` is how the multiple versions of the documentation site works. Looking at it again, there's really no reason to keep a redundant directory in `docs/_build` around anymore. I'll make a new issue to remove this.

jtgrasb, 2025-01-30:
Resolved by #387.

PR #387 (2025-01-30): Remove redundant Sphinx build directory
## Description
Closes #376 and #380 

## Type of PR
- [ ] Bug fix
- [ ] New feature
- [ ] Documentation
- [x] Other: **Cleanup**

## Checklist for PR
- [x] Authors read the [contribution guidelines](https://github.com/sandialabs/WecOptTool/blob/main/.github/CONTRIBUTING.md)
- [x] The pull request is from an issue branch (not main) on *your* fork, to the [dev branch in WecOptTool](https://github.com/sandialabs/WecOptTool/tree/dev).
- [x] The authors have given the admins edit access
- [x] All changes adhere to the style guide including PEP8, Docstrings, a [... cut]
closed 2025-01-30 by jtgrasb