sandialabs/WecOptTool #399: Inertia matrix on immersed part only?

mancellin, 2025-03-18:
I'm not sure about computing the inertia matrix on the immersed part:

https://github.com/sandialabs/WecOptTool/blob/cde6b775f3ae5248b0b71e13c0e3797cf6a0cf3b/wecopttool/core.py#L2180-L2188

The shape above the free surface should affects the inertia matrix.

jtgrasb, 2025-04-17:
Thinking about it more with @cmichelenstrofer, Capytaine's `compute_rigid_body_inertia()` function calculates the density as:
``` 
density = rho if self.mass is None else self.mass/volume
```
So, I think we can handle it on the WecOptTool side. The approach would be to define the mass in WecOptTool based on the immersed part, then pass in the full mesh to `compute_rigid_body_inertia()` which will use the mass/volume as the density to calculate the correct inertia values.

jtgrasb, 2025-04-22:
Reopened according to comments in #417. Will need to create new PR to update accordingly.

PR #417 (2025-04-17): Inertia matrix fix
## Description
Capytaine's `compute_rigid_body_inertia()` function calculates the density as:
``` 
density = rho if self.mass is None else self.mass/volume
```
So, I think we can handle it on the WecOptTool side. The approach would be to define the mass in WecOptTool based on the immersed part, then pass in the full mesh to `compute_rigid_body_inertia()` which will use the mass/volume as the density to calculate the correct inertia values.

Resolves #399 

I've updated the process and warning messages to reflect this.